pimoroni/bme280-python #3: Add temperature offset function

sandyjmacdonald, 2019-11-05:
This would allow you to correct the temperature and in turn get more sensible relative humidity values.

Gadgetoid, 2019-11-05:
There's an example of this being implemented before for the BME680 here - https://github.com/pimoroni/bme680-python/issues/11

The BME280 solution should closely reflect this.

sandyjmacdonald, 2019-11-06:
Using an approximation to calculate dew point, and in turn using that to recalculate the relative humidity seems to work well in my testing. This could be added as an `adjusted_humidity()` function, to which a temperature offset value could be passed.

/Users/sandyjmacdonald/Downloads/Screenshot 2019-11-05 at 22.12.35.png
(from https://en.wikipedia.org/wiki/Dew_point)

Here's a couple of lines of code that worked for me:

```
dewpoint = temperature - ((100 - humidity) / 5)
corrected_humidity = 100 - (5 * (corrected_temperature - dewpoint))
```

closed 2023-01-20 by Gadgetoid