pimoroni/bme280-python #5: install.sh error

Neex101, 2020-02-10:
On Rasbian: 2019-09-26-raspbian-buster-lite

As per install instructions

```
sudo pip install pimoroni-bme280
git clone https://github.com/pimoroni/bme280-python
cd bme280-python
sudo ./install.sh
```

Error received:

```
...
Installing for Python 3..
Traceback (most recent call last):
  File "setup.py", line 26, in <module>
    from setuptools import setup
ModuleNotFoundError: No module named 'setuptools'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "setup.py", line 28, in <module>
    from distutils.core import setup
ModuleNotFoundError: No module named 'distutils.core'
Done!
```

Can't get any of the examples to work - suspect this is the root cause?

Gadgetoid, 2020-02-10:
Yikes- no `setuptools` *or* `distutils`? I guess I shouldn't be surprised.

I guess it couldn't hurt to add a `sudo python3 -m pip install --upgrade setuptools` into the installer. Albeit on Pi it might also need a `sudo apt install python3-pip` in the first place.

Neex101, 2020-02-10:
I tried the recommended build steps:
`sudo apt install python3-pip`
`sudo python3 -m pip install --upgrade setuptools`
... and then reran install.sh
Issue fixed - thanks!

closed 2020-02-26 by Gadgetoid