pimoroni/bme280-python #12: ImportError: No module named bme280 when running all-values.py

alphanumeric007, 2020-10-31:
Fresh install of the latest PiOS on a Pi Zero W.
pHat Stack with Breakout Garden Mini i2c SPI and Mini i2c
BME280, RV3028, LTR-599 in the mini i2c BG
0.96 SPI Color LCD in the mini i2c SPI BG

My BME280 breakout is giving me the following error when I run the all-values.py file. 
Either from thoony or via command line.

from bme280 import BME280
ImportError: No module named bme280

i2c is enabled and it is detected. 
I have run the following.
sudo pip install pimoroni-bme280 smbus
sudo pip3 install pimoroni-bme280 smbus
and also did the following
git clone https://github.com/pimoroni/bme280-python
cd bme280-python
sudo ./install.sh
and rebooted several times.
Tried two different BME280’s, one that was working before I reimaged my SD card, and one new one I just received today. 
Both give the above error.

daniel-rhoades, 2020-10-31:
Same for me.  Looks like the packaging for the latest version (0.1.0 released today) maybe an issue.

After installing the previous version: `pip install pimoroni-bme280==0.0.2` it works.

alphanumeric007, 2020-10-31:
I'll have to give that a try.

alphanumeric007, 2020-10-31:
That has got mine working, thanks again. =)

tedder, 2020-11-02:
The problem is the wheel files. They don't contain any code:
```
$ unzip -l pimoroni_bme280-0.1.0-py3-none-any.whl 
Archive:  pimoroni_bme280-0.1.0-py3-none-any.whl
  Length      Date    Time    Name
---------  ---------- -----   ----
     1070  10-30-2020 11:25   pimoroni_bme280-0.1.0.dist-info/LICENSE.txt
     2243  10-30-2020 11:25   pimoroni_bme280-0.1.0.dist-info/METADATA
       92  10-30-2020 11:25   pimoroni_bme280-0.1.0.dist-info/WHEEL
        1  10-30-2020 11:25   pimoroni_bme280-0.1.0.dist-info/top_level.txt
      429  10-30-2020 11:25   pimoroni_bme280-0.1.0.dist-info/RECORD
---------                     -------
     3835                     5 files
```

Gadgetoid, 2020-11-02:
@tedder ugh! Good spot. I had put the `package` directive in the wrong part of `setup.cfg` and it quietly broke the package.

rickybalboa63, 2021-01-15:
I still have this problem. If I run the program using IDLE python(3.7.3) it doesn't work but if I run it in the cmd terminal using 'python all-values.py' it does work. I have tried sudo pip3 install pimoroni-bme280 smbus and this does not solve it.
Any help please, trying to get this to work with matplotlib and that needs python3.
Thanks

Gadgetoid, 2021-03-15:
@rickybalboa63 if you haven't discovered so already, you need to be explicit about upgrading a package:

```
sudo pip3 install --upgrade pimoroni-bme280
```

And it sometimes pays to refer to the specific Python version since... well Python can get dicey relating `pip` back to the Python you expect something else to be running in:

```
python37 -m pip install --upgrade pimoroni-bme280
```

tin-, 2022-07-06:
This package seem to be still broken. Tried to run in on Pi3 in either Python2 or Python3 environment with same results:

```
root@raspi:/home/root/Pimoroni/pimoroni-bme280/examples# python ./all-values.py
Traceback (most recent call last):
  File "./all-values.py", line 8, in <module>
    from bme280 import BME280
ImportError: No module named 'bme280'
```

```
root@raspi:/home/root/Pimoroni/pimoroni-bme280/examples# python2 ./all-values.py
Traceback (most recent call last):
  File "./all-values.py", line 8, in <module>
    from bme280 import BME280
ImportError: cannot import name BME280
```

Device is present in i2c detect and package was installed using both github with latest code or via pip.
```

root@raspb:/home/root/Pimoroni/pimoroni-bme280/examples# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- 77
```

```
root@raspi:/home/root/Pimoroni/pimoroni-bme280/examples# uname -a
Linux raspb 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux
```

```
root@raspi:/home/root/Pimoroni/pimoroni-bme280/examples# python --version
Python 3.5.3
```

```
root@raspb:/home/root/Pimoroni/pimoroni-bme280/examples# python2 --version
Python 2.7.13
```

alphanumeric007, 2022-07-06:
Try running 
sudo pip3 install pimoroni-bme280 smbus
Then try the examples again.

Gadgetoid, 2023-11-08:
Fingers crossed I don't break everything again with v1.0.0...