pimoroni/bme280-python #20: Unable to find bme280 on 0x76, CHIP_ID returned 58

nsteinmetz, 2021-11-14:
Hi there,

Started playing with bme280 - I can't initialise it correctly on my RPI + Breakout Garden + BME280 (and a few others breakout)

```python
#!/usr/bin/env python3

# std lib imports
import os
import logging
import argparse
import sys
import time

# extra imports
try:
    from smbus2 import SMBus
except ImportError:
    from smbus import SMBus

from bme280 import BME280

# Set logs
logger = logging.getLogger('bme280')
logger.setLevel(logging.INFO)
# Log format
formatter = logging.Formatter('%(asctime)s - %(name)s - %(levelname)s - %(message)s')
# Console Handler
ch = logging.StreamHandler()
ch.setLevel(logging.INFO)
ch.setFormatter(formatter)
logger.addHandler(ch)

def temperature():
    # Initialise the BME280
    bus = SMBus(1)
    bme280 = BME280(i2c_dev=bus)

    temperature = bme280.get_temperature()
    pressure = bme280.get_pressure()
    humidity = bme280.get_humidity()
    print('{:05.2f}*C {:05.2f}hPa {:05.2f}%'.format(temperature, pressure, humidity))
    time.sleep(1)

def main():
    """
    parser = argparse.ArgumentParser()
    parser.add_argument('--url', default="http://localhost", help="url")
    args = parser.parse_args()
    """

    temperature()

if __name__ == "__main__":
    main()
```

Leads to:

```
Traceback (most recent call last):
  File "/home/pi/airborne/airborne/environment.py", line 52, in <module>
    main()
  File "/home/pi/airborne/airborne/environment.py", line 49, in main
    temperature()
  File "/home/pi/airborne/airborne/environment.py", line 36, in temperature
    temperature = bme280.get_temperature()
  File "/home/pi/.local/lib/python3.9/site-packages/bme280/__init__.py", line 264, in get_temperature
    self.update_sensor()
  File "/home/pi/.local/lib/python3.9/site-packages/bme280/__init__.py", line 250, in update_sensor
    self.setup()
  File "/home/pi/.local/lib/python3.9/site-packages/bme280/__init__.py", line 228, in setup
    raise RuntimeError("Unable to find bme280 on 0x{:02x}, CHIP_ID returned {:02x}".format(self._i2c_addr, chip.id))
RuntimeError: Unable to find bme280 on 0x76, CHIP_ID returned 58
```

whereas it's on 0X76:

```
i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: 10 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- 38 -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- 76 --                         
```
Any changes on the chip that should be updated in the lib ?

othy1010, 2022-12-24:
> Hi there,
> 
> Started playing with bme280 - I can't initialise it correctly on my RPI + Breakout Garden + BME280 (and a few others breakout)
> 
> ```python
> #!/usr/bin/env python3
> 
> # std lib imports
> import os
> import logging
> import argparse
> import sys
> import time
> 
> # extra imports
> try:
>     from smbus2 import SMBus
> except ImportError:
>     from smbus import SMBus
> 
> from bme280 import BME280
> 
> # Set logs
> logger = logging.getLogger('bme280')
> logger.setLevel(logging.INFO)
> # Log format
> formatter = logging.Formatter('%(asctime)s - %(name)s - %(levelname)s - %(message)s')
> # Console Handler
> ch = logging.StreamHandler()
> ch.setLevel(logging.INFO)
> ch.setFormatter(formatter)
> logger.addHandler(ch)
> 
> 
> def temperature():
>     # Initialise the BME280
>     bus = SMBus(1)
>     bme280 = BME280(i2c_dev=bus)
> 
>     temperature = bme280.get_temperature()
>     pressure = bme280.get_pressure()
>     humidity = bme280.get_humidity()
>     print('{:05.2f}*C {:05.2f}hPa {:05.2f}%'.format(temperature, pressure, humidity))
>     time.sleep(1)
> 
> def main():
>     """
>     parser = argparse.ArgumentParser()
>     parser.add_argument('--url', default="http://localhost", help="url")
>     args = parser.parse_args()
>     """
> 
>     temperature()
> 
> if __name__ == "__main__":
>     main()
> ```
> 
> Leads to:
> 
> ```
> Traceback (most recent call last):
>   File "/home/pi/airborne/airborne/environment.py", line 52, in <module>
>     main()
>   File "/home/pi/airborne/airborne/environment.py", line 49, in main
>     temperature()
>   File "/home/pi/airborne/airborne/environment.py", line 36, in temperature
>     temperature = bme280.get_temperature()
>   File "/home/pi/.local/lib/python3.9/site-packages/bme280/__init__.py", line 264, in get_temperature
>     self.update_sensor()
>   File "/home/pi/.local/lib/python3.9/site-packages/bme280/__init__.py", line 250, in update_sensor
>     self.setup()
>   File "/home/pi/.local/lib/python3.9/site-packages/bme280/__init__.py", line 228, in setup
>     raise RuntimeError("Unable to find bme280 on 0x{:02x}, CHIP_ID returned {:02x}".format(self._i2c_addr, chip.id))
> RuntimeError: Unable to find bme280 on 0x76, CHIP_ID returned 58
> ```
> 
> whereas it's on 0X76:
> 
> ```
> i2cdetect -y 1
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:                         -- -- -- -- -- -- -- -- 
> 10: 10 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 30: -- -- -- -- -- -- -- -- 38 -- -- -- -- -- -- -- 
> 40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 70: -- -- -- -- -- -- 76 --                         
> ```
> 
> Any changes on the chip that should be updated in the lib ?
> 
> Thanks

hello, did you solve this problem I'm facing the same thing, Thanks

Gadgetoid, 2023-11-08:
Chip ID `0x58` is for a BMP280 not a BME280, this library is not compatible with that device.

You'll need: https://github.com/pimoroni/bmp280-python

closed 2023-11-08 by Gadgetoid