pimoroni/bme280-python #25: Raspberry Pi hangs in forced mode

mbcoder, 2023-07-21:
I'm finding that when I'm using the library in forced mode after a number of sensor reads, the Raspberry Pi hangs.

Is there a known issue with repeatedly reading the sensor in forced mode.  I'm reading the sensor every minute, but after a while something is locking up the Raspberry Pi.

Gadgetoid, 2023-11-08:
Not super helpful, I know, but it seems to be running continuously for me here. Is it still locking up?

mbcoder, 2023-11-08:
Yes it does lock up - I never did resolve the issue.  

I'm using a Raspberry Pi 3b+ and a 64bit Debian OS.

Gadgetoid, 2023-11-08:
I've managed to dig up a 3B v1.2. Do you have a link to your OS image?

mbcoder, 2023-11-08:
Using the MacOS imaging software this is what I've been selecting 
[image]

Gadgetoid, 2023-11-10:
Okay, I've set up a Pi 3B running the same OS and the "all-values.py" example. I ran an overnight test on the Pi 5, but it was still running this morning.

I suspect there might be some other factor at play I'm not recreating here, though.