pimoroni/bme280-python #28: ImportError

ShorTie8, 2023-11-14:
Trying to run some examples, but not working real well
`pi@Grow4:~/bme280-python/examples $ ls
all-values.py               dump-calibration.py  relative-altitude.py    temperature-forced-mode.py
compensated-temperature.py  local_altitude.py    temperature-compare.py
pi@Grow4:~/bme280-python/examples $ python compensated-temperature.py
Traceback (most recent call last):
  File "/home/pi/bme280-python/examples/compensated-temperature.py", line 6, in <module>
    from bme280 import BME280
ImportError: cannot import name 'BME280' from 'bme280' (/usr/local/lib/python3.9/dist-packages/bme280/__init__.py)
pi@Grow4:~/bme280-python/examples $ python dump-calibration.py
Traceback (most recent call last):
  File "/home/pi/bme280-python/examples/dump-calibration.py", line 7, in <module>
    from bme280 import BME280
ImportError: cannot import name 'BME280' from 'bme280' (/usr/local/lib/python3.9/dist-packages/bme280/__init__.py)
pi@Grow4:~/bme280-python/examples $ python relative-altitude.py
Traceback (most recent call last):
  File "/home/pi/bme280-python/examples/relative-altitude.py", line 9, in <module>
    from bme280 import BME280
ImportError: cannot import name 'BME280' from 'bme280' (/usr/local/lib/python3.9/dist-packages/bme280/__init__.py)
pi@Grow4:~/bme280-python/examples $
`
Jeff

Gadgetoid, 2023-11-14:
How did you install the library?

Gadgetoid, 2023-11-15:
Did you `pip install bme280` instead of `pip install pimoroni-bme280`?

ShorTie8, 2023-11-16:
sudo pip install pimoroni-bme280
is what I did

ShorTie8, 2023-11-16:
also ran the installer too...

Gadgetoid, 2023-11-16:
Must be somehow getting an older version of the library, since there *was* a bug like this prior to v0.1.1 - https://github.com/pimoroni/bme280-python/commit/91071570e57bd0726a53f08b915090575c163e39

What do you see if you: `pip freeze | grep bme280`

And also: 

```
cat /usr/local/lib/python3.9/dist-packages/bme280/__init__.py
```

ShorTie8, 2023-11-16:
pi@Grow4:~/bme280-python/examples $
pi@Grow4:~/bme280-python/examples $
pi@Grow4:~/bme280-python/examples $ pip freeze | grep bme280
bme280==0.6
pimoroni-bme280==0.1.1
RPi.bme280==0.2.4
pi@Grow4:~/bme280-python/examples $ cat /usr/local/lib/python3.9/dist-packages/bme280/__init__.py
#!/usr/bin/env python
# -*- coding: utf-8 -*-

# The MIT License (MIT)
#
# Copyright (c) 2016 Richard Hull
#
# Permission is hereby granted, free of charge, to any person obtaining a copy
# of this software and associated documentation files (the "Software"), to deal
# in the Software without restriction, including without limitation the rights
# to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
# copies of the Software, and to permit persons to whom the Software is
# furnished to do so, subject to the following conditions:
#
# The above copyright notice and this permission notice shall be included in
# all copies or substantial portions of the Software.
#
# THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
# IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
# FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
# AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
# LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
# OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
# SOFTWARE.

"""
Raspberry Pi BME280 Driver.
"""

__version__ = "0.2.4"

import datetime
import time
import uuid

from bme280.reader import reader
import bme280.const as oversampling
import pytz

# Oversampling modes
oversampling.x1 = 1
oversampling.x2 = 2
oversampling.x4 = 3
oversampling.x8 = 4
oversampling.x16 = 5

DEFAULT_PORT = 0x76

class uncompensated_readings(object):

    def __init__(self, block):
        self._block = block
        self.pressure = (block[0] << 16 | block[1] << 8 | block[2]) >> 4
        self.temperature = (block[3] << 16 | block[4] << 8 | block[5]) >> 4
        self.humidity = block[6] << 8 | block[7]

    def __repr__(self):
        return "uncompensated_reading(temp=0x{0:08X}, pressure=0x{1:08X}, humidity=0x{2:08X}, block={3})".format(
            self.temperature, self.pressure, self.humidity,
            ":".join("{0:02X}".format(c) for c in self._block))

class compensated_readings(object):
    """
[... 125 lines cut ...]
    Primes the sensor for reading (defaut: x1 oversampling), pauses for a set
    amount of time so that the reading stabilizes, and then returns a
    compensated reading object with the following attributes:
        * timestamp (Python's datetime object) when reading was taken.
        * temperature, in degrees Celcius.
        * humidity, in % relative humidity.
        * pressure, in hPa.
    """
    if compensation_params is None:
        compensation_params = __cache_calibration_params(bus, address)

    mode = 1  # forced
    t_oversampling = sampling or oversampling.x1
    h_oversampling = sampling or oversampling.x1
    p_oversampling = sampling or oversampling.x1

    bus.write_byte_data(address, 0xF2, h_oversampling)  # ctrl_hum
    bus.write_byte_data(address, 0xF4, t_oversampling << 5 | p_oversampling << 2 | mode)  # ctrl
    delay = __calc_delay(t_oversampling, h_oversampling, p_oversampling)
    time.sleep(delay)

    block = bus.read_i2c_block_data(address, 0xF7, 8)
    raw_data = uncompensated_readings(block)
    return compensated_readings(raw_data, compensation_params)
pi@Grow4:~/bme280-python/examples $

Gadgetoid, 2023-11-16:
Okay that bme280 library is not our bme280 library, I wonder if it's just shadowing ours or was installed by the installer accidentally.

Try:

```
pip uninstall bme280
pip install --upgrade pimoroni-bme280
```

ShorTie8, 2023-11-21:
Ok, I know there are Turtles and then Me.
Na, Been waiting on my pi5.
So it's fresh everything.
Like only history entries.

So, got me, still don't work

`
pi@pi5:~ $ history
    1  sudo apt update
    2  sudo apt upgrade
    3  sudo apt install python3-full
    4  sudo apt install i2c-tools
    5  sudo apt autoremove
pi@pi5:~ $ i2cdetect -a 1
70: -- -- -- -- -- -- 76 -- -- -- -- -- -- -- -- --

locate bme280.py
/usr/local/lib/python3.9/dist-packages/bme280/bme280.py
root@Grow4:/home/pi# ls -l /usr/local/lib/python3.9/dist-packages/bme28*
/usr/local/lib/python3.9/dist-packages/bme280:
total 40
-rw-r--r-- 1 root root 1037 Oct 16 01:40 bme280_i2c.py
-rw-r--r-- 1 root root 8393 Oct 16 01:40 bme280.py
-rw-r--r-- 1 root root 1699 Nov  5 00:50 const.py
-rw-r--r-- 1 root root 8357 Nov  5 00:50 __init__.py
drwxr-xr-x 2 root root 4096 Nov  5 00:50 __pycache__
-rw-r--r-- 1 root root 1971 Nov  5 00:50 reader.py

/usr/local/lib/python3.9/dist-packages/bme280-0.6.dist-info:
total 32
-rw-r--r-- 1 root root  337 Oct 16 01:40 DESCRIPTION.rst
-rw-r--r-- 1 root root   52 Oct 16 01:40 entry_points.txt
-rw-r--r-- 1 root root    4 Oct 16 01:40 INSTALLER
-rw-r--r-- 1 root root 1105 Oct 16 01:40 METADATA
-rw-r--r-- 1 root root 1079 Oct 16 01:40 metadata.json
-rw-r--r-- 1 root root 1184 Oct 16 01:40 RECORD
-rw-r--r-- 1 root root    0 Oct 16 01:40 REQUESTED
-rw-r--r-- 1 root root    7 Oct 16 01:40 top_level.txt
-rw-r--r-- 1 root root  110 Oct 16 01:40 WHEEL

pi@pi5:~ $ python3 -m pip install pimoroni-bme280
error: externally-managed-environment

× This environment is externally managed
?-> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.

`

Gadgetoid, 2023-11-22:
You should now be working in a virtual environment. I've explained why in exhaustive detail here: https://github.com/pimoroni/boilerplate-python/pull/13

If it sounds like tedious nonsense to you, welcome to the last two months of my life. 

In short, you'll need to:

```
sudo apt install python3-virtualenvwrapper virtualenvwrapper
source /usr/share/virtualenvwrapper/virtualenvwrapper.sh
mkvirtualenv your_project_name
```

Then install everything into that virtual environment with `pip install` and to get back into the environment:

```
source /usr/share/virtualenvwrapper/virtualenvwrapper.sh
workon your_project_name
```

You can add `source /usr/share/virtualenvwrapper/virtualenvwrapper.sh` to the bottom of `~/.bashrc` to avoid having to repeat that step every time.